Ardumower/ardumower #4: In robot.cpp: code unreachable in void Robot::checkBattery()

greymfm, 2015-04-01:
Revision name: STATE_OFF never attained when batVoltage < batSwitchOffIfBelow

Purpose of patch on this branch: Fix the issue.

--- robot.cpp	2014-10-24 09:44:38.000000000 +0200
+++ robot_0.1.cpp	2015-03-21 09:55:40.000000000 +0100
@@ -1325,14 +1325,14 @@
 // check (low) battery
 void Robot::checkBattery(){
   if (batMonitor){
-    if ((batVoltage < batGoHomeIfBelow) && (stateCurr !=STATE_OFF)) {
-      Console.println(F("triggered batGoHomeIfBelow"));
-      beep(2, true);      
-      setNextState(STATE_PERI_FIND, 0);
-    } else if ((batVoltage < batSwitchOffIfBelow) && (stateCurr !=STATE_OFF))  {
+    if ((batVoltage < batSwitchOffIfBelow) && (stateCurr !=STATE_OFF))  {
       Console.println(F("triggered batSwitchOffIfBelow"));
       beep(2, true);      
       setNextState(STATE_OFF, 0);
+    } else if ((batVoltage < batGoHomeIfBelow) && (stateCurr !=STATE_OFF)) {
+      Console.println(F("triggered batGoHomeIfBelow"));
+      beep(2, true);      
+      setNextState(STATE_PERI_FIND, 0);
     }
   }
 }

autega, 2015-04-21:
lets get this fixed

greymfm, 2015-04-21:
Yeah, fixes, fixes :)

autega, 2015-04-21:
hihi

closed 2015-05-02 by autega