Ardumower/ardumower #47: Serielle Konsole hängt sich auf

Kurzschuss, 2015-05-27:
habe gerade auf ein leeren Mega  die aktuelle Software aufgespielt:
im serielle Monitor bekomme ich folgende Meldung:
SETUP
ADCMan error: no calib data
matchSignal size=48
capture size=240
IMU error: no calib data
--------
accOfs=0.00,0.00,0.00
accScale=2.00,2.00,2.00
comOfs=0.00,0.00,0.00
comScale=2.00,2.00,2.00
--------
initL3G4200D

dann friert die anzeige ein
Hatte in der Mover Cpp alle Sensoren deaktiviert

das einfrieren war erst weg als ich den IMU angesteckt hatte.
wird evl trotz deaktivierten IMU beim Start der seriellen Konsolle nach dem IMU gesucht und auf antwort gewartet?

autega, 2015-06-05:
Kann ich nicht reproduzieren.  Ein nackter Mega mit dem aktuellen Code aus develop ohne irgendwelche Veränderungen in der mower.cpp läuft einwandfrei. Es wird natürlich gemeckert über fehlende Sensordaten, der Mega friert aber nicht ein.

closed 2015-06-05 by autega

FredericG-BE, 2015-06-05:
I had the same issue. I did not research the problem and to keep me going is commented the line
  imu.init(pinBuzzer);