Ardumower/ardumower #58: Support for ESP8266 WIFI module

FredericG-BE, 2015-07-08:
Ardumower implements the fpod protocol to configure, diagnose and control the Ardumower via an application on your Android phone or tablet.
 
The Ardumower PCB allows to plug in a Bluetooth module or a ESP8266 WIFI module to make the wireless connection between your Android device and your Ardumower. Currently only Bluetooth is supported by the Ardumower software.

FredericG-BE, 2015-07-08:
Work in progress:
 https://github.com/FredericG-BE/ardumower/wiki/Using-ESP8266-WIFI-module-on-Ardumower

greymfm, 2015-08-05:
This should be added to the Ardumower main code very soon - also, it will allow us to manually send commands to the mower (drive forward until obstacle at speed x, turn degree y, ...) and to get back sensor data (odometry, course, ...). This will allow us to more quickly develop SLAM algorithm code on the PC ('hardware-in-the-loop')...

greymfm, 2015-09-12:
ESP8266SerialWifiBridge feature: bulk transfers work faster with this ...
'''
 * HardwareSerial.h:
 *  #define SERIAL_TX_BUFFER_SIZE 4096
 *  #define SERIAL_RX_BUFFER_SIZE 4096
'''

'''
  //
  // What is received on the Serial Port is sent to the client
  //
  if (Serial.available()){
    size_t len = Serial.available();
    if (len != lenLastAvail) timeoutTime = millis() + 200;    
    lenLastAvail = len;
    if ((len >= 1500) || (millis() > timeoutTime)){
      uint8_t sbuf[len];
      Serial.readBytes(sbuf, len);      
      if (clientConnected){
        client.write(sbuf, len);
      }
      Serial.println(MSG_HEADER " OK");        
      lenLastAvail = 0;
    }
    // delay(1);    
  } 
'''

greymfm, 2015-09-15:
This code supports both normal AP client (channel=0), and soft-AP mode (if channel > 0, IP config is ignored):

```
example config:
   AP client:  config:0,SSID,PASSWD,192.168.2.14,192.168.2.1,255.255.255.0
   soft-AP mode: config:1,SSID,PASSWD,192.168.2.14,192.168.2.1,255.255.255.0
```

```
/*
 * HardwareSerial.h:
 *  #define SERIAL_TX_BUFFER_SIZE 4096
 *  #define SERIAL_RX_BUFFER_SIZE 4096
 * 
 */

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <Ticker.h>

#ifndef min
#define min(a,b) ((a)<(b)?(a):(b))
#endif // min

// When ECHOTEST is enabled, each received {xxx} will be echoed as [xxx]
// #define ECHOTEST
 
#define BAUDRATE 115200

#define MAX_CONFIG_LEN  100
#define MSG_HEADER "[WSB]"
#define VESRION "v0.2"
#define CONFIG_MSG_START "config:"

typedef struct {
  const char* name;
  const char* valueStr;
} param_t;

typedef struct ledSequence_t{
  uint8_t onTicks;
  uint8_t offTicks;
} ledSequence_t;

struct {
  boolean on;
  uint8_t counter;
  const ledSequence_t* currSequence;
} ledStatus;

char configMsg[MAX_CONFIG_LEN];
param_t params[] = { 
  {"Channel",""},  
  {"SSID",""},
  {"Password",""},
  {"IPAddress",""},
  {"Gateway",""},
  {"Subnet",""},
};
#define PARAMID_CHANNEL 0
#define PARAMID_SSID    1
#define PARAMID_PASSWD  2
#define PARAMID_LOCALIP 3
#define PARAMID_GATEWAY 4
#define PARAMID_SUBNET  5
#define NBPARAMS (sizeof(params)/sizeof(params[0]))

const ledSequence_t ledSeq_startup        =   {1,10};             
const ledSequence_t ledSeq_waitForConfig  =   {1,1};      
const ledSequence_t ledSeq_connecting  =      {3,3};      
[... 287 lines cut ...]
      }
    }
  }
  
  //
  // What is received on the Serial Port is sent to the client
  //
  if (Serial.available()){
    size_t len = min(4096, Serial.available());
    if (len != lenLastAvail) timeoutTime = millis() + 100;    
    lenLastAvail = len;
    if ((len >= 1500) || (millis() > timeoutTime)){
      uint8_t sbuf[len];
      Serial.readBytes(sbuf, len);      
      if (clientConnected){
        client.write(sbuf, len);
      }
      Serial.println(MSG_HEADER " OK");        
      lenLastAvail = 0;
    }
    // delay(1);    
  } 
}

```

greymfm, 2015-09-15:
Updated the code above: buffer is not allocated for each packet anymore. Seem to work more stable.

xjustbmw, 2019-09-28:
any news regarding this code with the new esp library that does not support changing the buffer size anymore ?

for me it barely works on mega using esp-link 
esp-link works good on due only