Ardumower/ardumower #60: Perimeter plot can freeze

FredericG-BE, 2015-07-11:
The problem is caused by the fact that ADCMan captures the perimeter samples and that there are 2 consumers for the data while not really supported by ADCMan

Perimeter will poll for ADCMan.isCaptureComplete(idxPin[idx]) for the perimeter signal to be available, will consume it and invalidate it via ADCMan.restart(idxPin[idx]); 
Pfod does that too. If periodically (but much much slower) polls for the capture to be available, but it has to be lucky that Perimeter has not invalidated it in the meantime. And when it has it takes >200ms for Perimeter to try it again. 
It can now happen that both get synchonized and Pfod has a very hard time getting the data. This can go one for minutes, sometimes the sychonization is broken and the plot goes on. 
It must be mentioned that I have this on a slight modified code (with less ADCs to monitor) and that this unfortunate synchronization does not happen in the released versions. However, this shows that any seemingly unrelated change can cause this mechanism to start failing.

Here a few scope plots. The blue line is high when the ADC data is available; the red line toggles when pfod tries to get the data. Here are examples where pfod is not able to get the data and this can go on for minutes

[image]
[image]

FredericG-BE, 2015-07-12:
From 12164f9dc4b339eb7c0bd6757e0e2faa2c3a42ab Mon Sep 17 00:00:00 2001
From: Frederic Goddeeris <frederic@rc-flight.be>
Date: Sun, 12 Jul 2015 20:26:16 +0200
Subject: [PATCH] Issue #61 Perimeter plot can freeze - let pfod get the signal
 sample from perimeter instead of directly via ADCMan

---
 ardumower/perimeter.cpp |  8 ++++++++
 ardumower/perimeter.h   |  5 ++++-
 ardumower/pfod.cpp      | 46 +++++++++++++++++++++++-----------------------
 ardumower/pfod.h        |  3 ++-
 4 files changed, 37 insertions(+), 25 deletions(-)

diff --git a/ardumower/perimeter.cpp b/ardumower/perimeter.cpp
index 58d7e80..17677a5 100644
--- a/ardumower/perimeter.cpp
+++ b/ardumower/perimeter.cpp
@@ -98,8 +98,16 @@ void Perimeter::speedTest(){
   Console.println(loops);
 }
 
+const int8_t* Perimeter::getRawSignalSample(byte idx) {
+  return rawSignalSample[idx];
+}
+
 int Perimeter::getMagnitude(byte idx){  
   if (ADCMan.isCaptureComplete(idxPin[idx])) {
+    // Keep a sample of the raw signal
+    memset(rawSignalSample[idx], 0, RAW_SIGNAL_SAMPLE_SIZE);
+    memcpy(rawSignalSample[idx], ADCMan.getCapture(idxPin[idx]), min(ADCMan.getCaptureSize(idxPin[0]), RAW_SIGNAL_SAMPLE_SIZE));
+    // Process signal
     matchedFilter(idx);
   }
   return mag[idx];
diff --git a/ardumower/perimeter.h b/ardumower/perimeter.h
index 510d0d4..4c51b78 100644
--- a/ardumower/perimeter.h
+++ b/ardumower/perimeter.h
@@ -37,6 +37,7 @@ How to use it (example):
 
 #include <Arduino.h>
 
+#define RAW_SIGNAL_SAMPLE_SIZE 32
 
 
 class Perimeter
@@ -44,7 +45,8 @@ class Perimeter
   public:
     Perimeter();
     // set ADC pins
-    void setPins(byte idx0Pin, byte idx1Pin);    
+    void setPins(byte idx0Pin, byte idx1Pin);
+    const int8_t* getRawSignalSample(byte idx);
     // get perimeter magnitude
     int getMagnitude(byte idx);    
     int getSmoothMagnitude(byte idx);
@@ -75,6 +77,7 @@ class Perimeter
     int16_t signalMax[2];
     int16_t signalAvg[2];    
     int signalCounter[2];    
+    int8_t rawSignalSample[2][RAW_SIGNAL_SAMPLE_SIZE];
     void matchedFilter(byte idx);
     int16_t corrFilter(int8_t *H, int8_t subsample, int16_t M, int8_t *ip, int16_t nPts, float &quality);
     void printADCMinMax(int8_t *samples);
diff --git a/ardumower/pfod.cpp b/ardumower/pfod.cpp
index b672547..07c947d 100644
--- a/ardumower/pfod.cpp
+++ b/ardumower/pfod.cpp
@@ -1262,30 +1262,30 @@ void RemoteControl::run(){
     }
   } else if (pfodState == PFOD_PLOT_PERIMETER){    
     if (millis() >= nextPlotTime){
-      if (perimeterCaptureIdx == 32*3){
-        if (ADCMan.isCaptureComplete(A5)){
-          int8_t *samples = ADCMan.getCapture(A5);      
-          memcpy(perimeterCapture, samples, 32);
-          perimeterCaptureIdx = 0;
-        }
-      }
-      if (perimeterCaptureIdx < 32*3){      
-        nextPlotTime = millis() + 200;            
-        Bluetooth.print(perimeterCapture[perimeterCaptureIdx / 3]);
-        Bluetooth.print(",");
-        Bluetooth.print(robot->perimeterMag);
-        Bluetooth.print(",");
-        Bluetooth.print(robot->perimeter.getSmoothMagnitude(0));
-        Bluetooth.print(",");
-        Bluetooth.print(robot->perimeter.isInside(0));
-        Bluetooth.print(",");
-        Bluetooth.print(robot->perimeterCounter);
-        Bluetooth.print(",");
-        Bluetooth.print(!robot->perimeter.signalTimedOut(0));
-        Bluetooth.print(",");
-        Bluetooth.println(robot->perimeter.getFilterQuality(0));
-        perimeterCaptureIdx++;
+
+      if (perimeterCaptureIdx>=RAW_SIGNAL_SAMPLE_SIZE*3)
+              perimeterCaptureIdx = 0;
+
+      if (perimeterCaptureIdx == 0) {
+        // Get new Perimeter sample to plot
+        memcpy(perimeterCapture, robot->perimeter.getRawSignalSample(0), RAW_SIGNAL_SAMPLE_SIZE);
       }
+
+      nextPlotTime = millis() + 200;
+      Bluetooth.print(perimeterCapture[perimeterCaptureIdx / 3]);
+      Bluetooth.print(",");
+      Bluetooth.print(robot->perimeterMag);
+      Bluetooth.print(",");
+      Bluetooth.print(robot->perimeter.getSmoothMagnitude(0));
+      Bluetooth.print(",");
+      Bluetooth.print(robot->perimeter.isInside(0));
+      Bluetooth.print(",");
+      Bluetooth.print(robot->perimeterCounter);
+      Bluetooth.print(",");
+      Bluetooth.print(!robot->perimeter.signalTimedOut(0));
+      Bluetooth.print(",");
+      Bluetooth.println(robot->perimeter.getFilterQuality(0));
+      perimeterCaptureIdx++;
     }
   } else if (pfodState == PFOD_PLOT_GPS){
     if (millis() >= nextPlotTime){
diff --git a/ardumower/pfod.h b/ardumower/pfod.h
index 220e1f9..f5c5c27 100644
--- a/ardumower/pfod.h
+++ b/ardumower/pfod.h
@@ -40,6 +40,7 @@
 #include <Arduino.h>
 #include "drivers.h"
 #include "pid.h"
+#include "perimeter.h"
 
 // pfodApp state
 enum { PFOD_OFF, PFOD_MENU, PFOD_LOG_SENSORS, 
@@ -65,7 +66,7 @@ class RemoteControl
     byte pfodState;
     int testmode;
     unsigned long nextPlotTime;
-    int8_t perimeterCapture[32]; 
+    int8_t perimeterCapture[RAW_SIGNAL_SAMPLE_SIZE];
     int perimeterCaptureIdx;        
     float stringToFloat(String &s);   
 
-- 
1.9.5.msysgit.0

greymfm, 2015-07-12:
What about this code? The ADC buffer filling get's a chance in each iteration then...

</pre>
  } else if (pfodState == PFOD_PLOT_PERIMETER){    
    if (perimeterCaptureIdx == 32*3){
      if (ADCMan.isCaptureComplete(A5)){
      int8_t *samples = ADCMan.getCapture(A5);      
      memcpy(perimeterCapture, samples, 32);
      perimeterCaptureIdx = 0;
    }
    if (millis() >= nextPlotTime){      
      if (perimeterCaptureIdx < 32*3){      
        nextPlotTime = millis() + 200;            
        Bluetooth.print(perimeterCapture[perimeterCaptureIdx / 3]);          
        Bluetooth.print(",");                    
        Bluetooth.print(robot->perimeterMag);
        Bluetooth.print(",");
        Bluetooth.print(robot->perimeter.getSmoothMagnitude(0));
        Bluetooth.print(",");
        Bluetooth.print(robot->perimeter.isInside(0));
        Bluetooth.print(",");
        Bluetooth.print(robot->perimeterCounter);
        Bluetooth.print(",");        
        Bluetooth.print(!robot->perimeter.signalTimedOut(0));        
        Bluetooth.print(",");                
        Bluetooth.println(robot->perimeter.getFilterQuality(0));                
        perimeterCaptureIdx++;
      }
    }
</pre>

FredericG-BE, 2015-07-12:
This is the original code right? 
When isCaptureComplete() returns false, nextPlotTime is not being increased, but in any case it takes another 200ms before the code will be called again. When I print perimeterCaptureIdx it counts to 96 and then gets stuck there for minutes.
It is funny it is not a common problem, because I have it all the time...

greymfm, 2015-07-12:
a patch was submitted:

```
From 12164f9dc4b339eb7c0bd6757e0e2faa2c3a42ab Mon Sep 17 00:00:00 2001
From: Frederic Goddeeris <frederic@rc-flight.be>
Date: Sun, 12 Jul 2015 20:26:16 +0200
Subject: [PATCH] Issue #61 Perimeter plot can freeze - let pfod get the signal
 sample from perimeter instead of directly via ADCMan

---
 ardumower/perimeter.cpp |  8 ++++++++
 ardumower/perimeter.h   |  5 ++++-
 ardumower/pfod.cpp      | 46 +++++++++++++++++++++++-----------------------
 ardumower/pfod.h        |  3 ++-
 4 files changed, 37 insertions(+), 25 deletions(-)

diff --git a/ardumower/perimeter.cpp b/ardumower/perimeter.cpp
index 58d7e80..17677a5 100644
--- a/ardumower/perimeter.cpp
+++ b/ardumower/perimeter.cpp
@@ -98,8 +98,16 @@ void Perimeter::speedTest(){
   Console.println(loops);
 }
 
+const int8_t* Perimeter::getRawSignalSample(byte idx) {
+  return rawSignalSample[idx];
+}
+
 int Perimeter::getMagnitude(byte idx){  
   if (ADCMan.isCaptureComplete(idxPin[idx])) {
+    // Keep a sample of the raw signal
+    memset(rawSignalSample[idx], 0, RAW_SIGNAL_SAMPLE_SIZE);
+    memcpy(rawSignalSample[idx], ADCMan.getCapture(idxPin[idx]), min(ADCMan.getCaptureSize(idxPin[0]), RAW_SIGNAL_SAMPLE_SIZE));
+    // Process signal
     matchedFilter(idx);
   }
   return mag[idx];
diff --git a/ardumower/perimeter.h b/ardumower/perimeter.h
index 510d0d4..4c51b78 100644
--- a/ardumower/perimeter.h
+++ b/ardumower/perimeter.h
@@ -37,6 +37,7 @@ How to use it (example):
 
 #include <Arduino.h>
 
+#define RAW_SIGNAL_SAMPLE_SIZE 32
 
 
 class Perimeter
@@ -44,7 +45,8 @@ class Perimeter
   public:
     Perimeter();
     // set ADC pins
-    void setPins(byte idx0Pin, byte idx1Pin);    
+    void setPins(byte idx0Pin, byte idx1Pin);
+    const int8_t* getRawSignalSample(byte idx);
     // get perimeter magnitude
     int getMagnitude(byte idx);    
     int getSmoothMagnitude(byte idx);
@@ -75,6 +77,7 @@ class Perimeter
     int16_t signalMax[2];
     int16_t signalAvg[2];    
     int signalCounter[2];    
+    int8_t rawSignalSample[2][RAW_SIGNAL_SAMPLE_SIZE];
     void matchedFilter(byte idx);
     int16_t corrFilter(int8_t *H, int8_t subsample, int16_t M, int8_t *ip, int16_t nPts, float &quality);
     void printADCMinMax(int8_t *samples);
diff --git a/ardumower/pfod.cpp b/ardumower/pfod.cpp
index b672547..07c947d 100644
--- a/ardumower/pfod.cpp
+++ b/ardumower/pfod.cpp
@@ -1262,30 +1262,30 @@ void RemoteControl::run(){
     }
   } else if (pfodState == PFOD_PLOT_PERIMETER){    
     if (millis() >= nextPlotTime){
-      if (perimeterCaptureIdx == 32*3){
-        if (ADCMan.isCaptureComplete(A5)){
-          int8_t *samples = ADCMan.getCapture(A5);      
-          memcpy(perimeterCapture, samples, 32);
-          perimeterCaptureIdx = 0;
-        }
-      }
-      if (perimeterCaptureIdx < 32*3){      
-        nextPlotTime = millis() + 200;            
-        Bluetooth.print(perimeterCapture[perimeterCaptureIdx / 3]);
-        Bluetooth.print(",");
-        Bluetooth.print(robot->perimeterMag);
-        Bluetooth.print(",");
-        Bluetooth.print(robot->perimeter.getSmoothMagnitude(0));
-        Bluetooth.print(",");
-        Bluetooth.print(robot->perimeter.isInside(0));
-        Bluetooth.print(",");
-        Bluetooth.print(robot->perimeterCounter);
-        Bluetooth.print(",");
-        Bluetooth.print(!robot->perimeter.signalTimedOut(0));
-        Bluetooth.print(",");
-        Bluetooth.println(robot->perimeter.getFilterQuality(0));
-        perimeterCaptureIdx++;
+
+      if (perimeterCaptureIdx>=RAW_SIGNAL_SAMPLE_SIZE*3)
+              perimeterCaptureIdx = 0;
+
+      if (perimeterCaptureIdx == 0) {
+        // Get new Perimeter sample to plot
+        memcpy(perimeterCapture, robot->perimeter.getRawSignalSample(0), RAW_SIGNAL_SAMPLE_SIZE);
       }
+
+      nextPlotTime = millis() + 200;
+      Bluetooth.print(perimeterCapture[perimeterCaptureIdx / 3]);
+      Bluetooth.print(",");
+      Bluetooth.print(robot->perimeterMag);
+      Bluetooth.print(",");
+      Bluetooth.print(robot->perimeter.getSmoothMagnitude(0));
+      Bluetooth.print(",");
+      Bluetooth.print(robot->perimeter.isInside(0));
+      Bluetooth.print(",");
+      Bluetooth.print(robot->perimeterCounter);
+      Bluetooth.print(",");
+      Bluetooth.print(!robot->perimeter.signalTimedOut(0));
+      Bluetooth.print(",");
+      Bluetooth.println(robot->perimeter.getFilterQuality(0));
+      perimeterCaptureIdx++;
     }
   } else if (pfodState == PFOD_PLOT_GPS){
     if (millis() >= nextPlotTime){
diff --git a/ardumower/pfod.h b/ardumower/pfod.h
index 220e1f9..f5c5c27 100644
--- a/ardumower/pfod.h
+++ b/ardumower/pfod.h
@@ -40,6 +40,7 @@
 #include <Arduino.h>
 #include "drivers.h"
 #include "pid.h"
+#include "perimeter.h"
 
 // pfodApp state
 enum { PFOD_OFF, PFOD_MENU, PFOD_LOG_SENSORS, 
@@ -65,7 +66,7 @@ class RemoteControl
     byte pfodState;
     int testmode;
     unsigned long nextPlotTime;
-    int8_t perimeterCapture[32]; 
+    int8_t perimeterCapture[RAW_SIGNAL_SAMPLE_SIZE];
     int perimeterCaptureIdx;        
     float stringToFloat(String &s);   
 
-- 
1.9.5.msysgit.0
```

greymfm, 2015-07-12:
Quote "This is the original code right?"

No, it's not the original code - I just moved the buffer filling out of the 200 ms loop. So the buffer could be filled every iteration, simply just after the ADC buffer was filled... Read the code and you'll understand :-)

FredericG-BE, 2015-07-12:
Sorry, yes I see the difference...
The thing is is that all this code is also only called every 200 ms:

  if (millis() >= nextTimePfodLoop){
    nextTimePfodLoop = millis() + 200;
    rc.run();        
  }

FredericG-BE, 2015-07-22:
Fixed in develop. Will be in > 1.0.4a

closed 2015-07-22 by FredericG-BE