Ardumower/ardumower #74: Wrong day of week validation in drivers.cpp

manolog, 2015-09-13:
Line 275 reads:  if (    (r.time.minute > 59) || (r.time.hour > 23) || (r.date.dayOfWeek > 6) 
    should read:   if (    (r.time.minute > 59) || (r.time.hour > 23) || (r.date.dayOfWeek > 7) 

as 7 is a valid week day number.

greymfm, 2015-09-14:
That would explain why the DS1307 sometimes is in invalid state :-) Datasheets says range is 1-7.  If I understand this correctly, the value cannot be zero (0) and we should check this.
http://datasheets.maximintegrated.com/en/ds/DS1307.pdf

So the range cannot be 0-6.

" The day-of-week register increments at midnight. Values that correspond to the day of week are user-defined but must be sequential (i.e., if 1 equals Sunday, then 2 equals Monday, and so on.) Illogical time and date entries result in undefined operation"

```readDS1307:
r.date.dayOfWeek     -= 1;  // day of week name array starts at zero
```

manolog, 2015-09-14:
I should have noticed earlier that the error always happens on Sundays right after turning back on the machine. Muy ardumower sleeps at home and I always push down the emergency switch at night in order to avoid surprises :)

Good point about correcting the array index; I always manage to offer half-cooked solutions to bugs.

Holoratte, 2015-09-15:
readDS1307:
r.date.dayOfWeek     -= 1;  // day of week name array starts at zero
...
dt = r;

Then line 298 (setDS1307) should be?
  buf[3] = dt.date.dayOfWeek + 1;

greymfm, 2016-05-29:
fixed in https://github.com/Ardumower/ardumower/commit/5c89591f152d81bfa57cd99239fc30f93b4a1cbf

closed 2021-03-14 by netlars