Ardumower/ardumower #90: License is non-Open Source

ericschultz, 2016-07-10:
The license of the software is currently non-open source. In particular, the license states that commercial use is not allowed which seems to be added as an exception to the GPLv3.  Commercial use must be allowed by the open source definition and the free software definition. Please remove the commercial use exception on the license or, at the very least, stop referring to this project as open source.

greymfm, 2016-07-10:
What we actually wanted with this is to offer two licenses:

**Open source license**

The open source license is designed for you to use Ardumower to build open source and personal projects. The Ardumower open source license is GPLv3. The GPLv3 has many terms, but the most important is how it is sticky when you distribute your work publicly. From the GPL FAQ:

    If you release the modified version to the public in some way, the GPL requires you to make the modified source code available to the program's users, under the GPL.

Releasing your project that uses Ardumower under the GPLv3, in turn, requires your project to be licensed under the GPLv3. If you are okay with this, feel free to use Ardumower under the GPLv3, without purchasing a commercial license.

**Commercial license**

The commercial license is designed to for you to use Ardumower in commercial products and applications, without the provisions of the GPLv3. With the commercial license, your code is kept propietary, to yourself. If you want to use Ardumower to develop commercial sites, themes, projects, and applications, the commercial license is the appropriate license.

ericschultz, 2016-07-10:
It would help if you clarified this in source files to say that you can alternatively license under a proprietary non-GPLv3 license if you'd prefer to not license under the GPLv3.

As an aside, you may need to be careful about the files you include in this repo. There are some files which aren't under a GPL compatible license (or even an open source license) (https://github.com/Ardumower/ardumower/blob/master/code/ardumower/flash_efc.h)