Ardumower/ardumower #115: Mow is ON/OFF should be a real switch

greymfm, 2017-06-14:
Mower should not turn on mowing motor if 'Mow is OFF' was activated

greymfm, 2017-06-16:
Wir könnten mit der 1.3 noch einen Schalter bei MOW in der APP einführen mit dem wir die Drehrichtung des Mow-Motors einstellen. Das hätte den Vorteil das wir die Klingen beidseitig abnutzen können ohne diese vom Teller entfernen zu müssen wenn sie auf einer Seite stumpf werden. Hat sonst keiner zumindes weiß ich davon nichts

Kurzschuss, 2017-06-16:
den Schalter würde ich machen aber so das wenn der Schalter aktiv ist
jedesmal wenn der Mähteller neu startet die Drehrichtung umgekehrt wird.
Das würde dazu führen das das Messer gleichmäßig von beiden Seiten
abgenutzt wird

greymfm, 2017-06-16:
Sehr gute Idee :-)