Ardumower/ardumower #123: Perimerter wir ignoriert, wenn Bumper/Sonar aktiv ist

surfingmicha, 2017-08-06:
Wird der Perimeter mit aktiver Bumper/Sonar- Erkennung überfahren, fährt der Mower nur kurz zurück und dann vollgas über den Perimeter.
Grun hierfür ist eine fehlende Fallbehandlung:

in robot.cpp -> void Robot::checkPerimeterBoundary(){

statt : else if ((stateCurr == STATE_ROLL)) {

muss STATE_REVERSE mit rein:

else if ((stateCurr == STATE_ROLL) || (stateCurr == STATE_REVERSE) ) {

netlars, 2021-03-27:
Hat das wirklich einen Effekt gebracht? Ich konnte keinen Verbesserung mit dem Part feststellen.
Dieser Commit hat eine merkliche Verbesserung bei mir gebracht 5176f575f5a32a0951bf62f642757d2c4642b6ca

netlars, 2021-03-29:
Beide hat keine Lösung gebracht.
Schon komisch, beim Rückwärtsfahren nach Perimeter Out, stoppt er und Vollgas nach vorn, Sonar wird eins Höher gezählt. Aber im Aufruf Perimeter_Out_Rev bzw Perimeter_Out_Roll gibt es gar kein CheckSonar, warum reagiert der dann auf Sonar?