Ardumower/ardumower #139: PCB 1.4

Kralg, 2021-05-19:
Hello, 

I would like to ask if there are any plans to upload here the PCB 1.4 kicad project. I would like to make some adjustments so it would be great to edit the most current version.  If I am not mistaken, do you @Kurzschuss know anything about this?

Kurzschuss, 2021-05-19:
Ja die Platine stammt von mir. Die Pläne sind hier zu finden:
https://github.com/Ardumower/ardumower/tree/master/pcb/Test_und_Entwicklung

Ich werde demnächst noch den Plan bereinigen und die interne Kommentare
entfernen:Am Schaltplan wird sich nichts ändern.
Wenn ich die Bereinigung durchgeführt habe ist der nbeue Plan dann hier zu
finden:
https://github.com/Ardumower/ardumower/tree/master/pcb/Produzierte_Platinen

vorabversion der Anleitung habe ich auch hochgeladen
Aber Achtung kann noch Fehler enthalten.
https://github.com/Ardumower/ardumower/tree/master/Dokumentation/Ardumower%20Platine%20V1.4%20Aufbauanleitung

Gruß Uwe (Kurzschuss)

Yes the circuit board is mine. The plans can be found here:
https://github.com/Ardumower/ardumower/tree/master/pcb/Test_und_Entwicklung

I will soon clean up the plan and remove the internal comments: Nothing
will change in the circuit diagram.
When I have done the cleanup, the new plan can be found here:
https://github.com/Ardumower/ardumower/tree/master/pcb/Produzierte_Platinen

I have also uploaded a preliminary version of the manual.
But beware, it may still contain errors.
https://github.com/Ardumower/ardumower/tree/master/Dokumentation/Ardumower%20Platine%20V1.4%20Aufbauanleitung

Greetings Uwe (Kurzschuss)

Translated with www.DeepL.com/Translator (free version)

Kralg, 2021-05-19:
Danke für die schnelle Antwort. Ich werde es mir ansehen. 
Tolles Board.

Kralg, 2021-05-20:
Hallo, Uwe, 

Können Sie die Dateien im Kicad-Format speichern? Ich kann .kicad_pro nicht öffnen

Kurzschuss, 2021-05-20:
Ich arbeit bei Kicad nicht mit der normalen Version sondern mit der
Entwicklerversion.
Ich vermute das es daran liegt bei ihnen
Die Entwickerversion ist hier zu finden:

https://kicad-downloads.s3.cern.ch/index.html?prefix=windows/nightly/

Gruß
 uwe

Kralg, 2021-05-20:
Ich habe gerade die Kicad von dem Link heruntergeladen, den Sie gesendet haben. Die Version: 
2021-05-20T08:02:58.104Z        1.0 GB         kicad-msvc.r22471.4d227d2d2b-x86_64.exe
Wenn ich das Programm starte und Datei-> Projekt öffnen wähle, wird dieses Fenster nur mit der Filterung von Dateien vom Typ .pro angezeigt. Können Sie welche Version verwenden, können Sie dort auch .kicad_pro auswählen oder wie öffnen Sie das Projekt? Ich habe gelesen, dass es Vorlagen für das Projekt geben kann. Verwenden Sie einen von ihnen? Wenn ja, können Sie bitte beschreiben, wie Sie das Programm so konfigurieren, dass dieses Projekt geöffnet werden kann? 
[image]

Kurzschuss, 2021-05-20:
Die Vorlagen brauchen sie nur wenn sie evl ein neues Projekt erstellen.
gehen sie bitte mal in den Datei explorer und wechseln sie in den Ordner wo
sie das Projekt heruntergeladen haben.
Kontrollieren sie da bitte mal ob alle Daten vorhanden sind.

eine alternative ist Github zu installieren und das ganze Ardumower Projekt
zu klonen.
Das zwar etwas Platz darin ist dann aber alles vorhanden

Der dateipfad sollte etwas so aussehen
siehe Bild
evl habe sie die dateien nicht richtig heruntergeladen und es fehlt etwas?
In Github verwend ich aich noch eigene Anpassungen die werden aber
hauptsächlich für die 3D Ansicht benötigt:
https://github.com/Ardumower/ardumower/tree/master/pcb/KiCad

https://forum.ardumower.de/threads/kicad.22955/
Achtung die Dateipfade haben sich etwas geändert

i
*ch habe im Dateianhang das Ardumower Projekt gepackt nochmal hinzugefügt*

*das hat leider nicht geklappt *

*deswegen ein Link in Google Drive von
mirhttps://drive.google.com/file/d/19za8Fmv5QyyQAdqyGV1KjrWTyx3N9sFY/view?usp=sharing
<https://drive.google.com/file/d/19za8Fmv5QyyQAdqyGV1KjrWTyx3N9sFY/view?usp=sharing>*

damit evl noch mal testen

Ansonsten noch mal melden
Sie sprechen Deutsch?

Gruß
Uwe [image: 2021-05-20 22_08_56-Window.jpg]

Kralg, 2021-05-21:
Danke für die Antwort. Ich spreche nur ein bisschen Deutsch, was für technische Details nicht ausreicht.
Mein Fehler, ich habe nicht gesehen, dass die Entwicklungs-Kicad etwas vollständig / zusätzliches Startsymbol hat. Jetzt kann ich auch .kicad_pro Dateien öffnen. Vielen Dank!
Ich werde mir die Beschreibung ansehen.

Kralg, 2021-05-23:
Hallo Uwe,

Ich habe die Stückliste überprüft und habe einige Fragen. Können Sie bitte schreiben, warum die erste Position nur BY3, BY5-B8 ist? Wenn ich bom generiere, kann ich sehen, dass es insgesamt 12 Positionen gibt.
In der 7. Zeile kann ich D23 im Projekt nicht finden, aber es könnte D12-D17-Positionen geben?
Kann es in der 8. Zeile sein, dass die Teile D4, D5 und nicht D4, D74 sind?
Können Sie bitte schreiben, ob diese Stückliste die neueste Version ist?

Kurzschuss, 2021-05-25:
Woher haben sie die Liste wo D23 vorkommt?   evl vo hier?
Ardumower_PCB1.4_parts_JLCPCB.ods  ?
Kann ignoriert die Tabelle war eine vorabversion.

Die Relevante Liste ist diese : *Bauteilliste JLCB.ods*
Das sind alle Bauteile die in China bestückt worden sind mit den
entsprechenden LCSC Part Nummern.
Diese Liste wurde direkt für den Produktionsauftrag verwendet worden

In dieser Datei sind nur die Bauteile die bestückt werden von den
Platinenhersteller und nicht die Bauteile die von Hand nachgearbeitet
werden müssen.

Die Dateien in den Gerber Ordner sind die Dateien wonach die Platine
gefertigt worden ist.

Gruß
  Uwe

Kralg, 2021-06-03:
Danke für die Antwort. Ich möchte nach UART fragen, ich plane, GPS und eines der Wlan oder Bluetooth für die Fernbedienung mit dem Smartphone zu verwenden. Ich möchte zwei weitere uarts verwenden, um Motortreiber zu steuern. In diesem Fall könnte ich UART0 und einen der UART1 oder UART2 verwenden. Können Sie schreiben bitte, wenn Sie irgendwelche Probleme für das sehen könnte?

Kurzschuss, 2021-06-04:
Das ist eine Frage die eigentlich mehr die Software betrifft. Da mal bitte
im Forum nachfragen. Soviel ich weiß wird bei unserer Platine der Uart0
nicht verwendet weil die schnittstelle den eigentlichen USB Anschluß
entspricht worüber der Arduino Programmiert wird. Welche genauen Probleme
dabei auftreten können kann ich leider nicht sagen. Das wäre ein Thema für
jemanden der sich mit der Software besser auskennt. ICh habe mich da auf
die Aussage verlassen die mir unserer Software entwickler gegeben hat weil
ich auch schon eine ähnlichen Vorschlag mit der Uart0 schnittstelle gemacht
habe. evl wäre es noch eine Möglichkeit 2 freie Pins als weitere
Schnittstelle zu definieren Inwieweit das möglich ist und geht bin ich
leider überfragt weil ich von der Programmierung zu wenig ahnung habe. evl
könnte man noch auf ein Bus System setzen.  Das kommt aber darauf an
inwieweit die Motortreiber das unterstützen. Wenn eigene Motortreiber
entwickelt werden sollten könnte man da evl ein Bussystem mit vorsehen.

Gruß
Uwe
Gruß
 Uwe

Kralg, 2021-06-07:
Hi, Uwe,
Danke fur die Antwort.
Ich habe an das Forum geschrieben, aber können Sie vorschlagen, wer diese Frage am besten beantworten könnte?

Kurzschuss, 2021-06-07:
Alexander macht die Programmierung

Gruß
 Uwe

closed 2021-08-17 by Kralg

woody-s, 2025-04-13:
Hallo Uwe,
ich bin neu im Thema Ardumower und suche bisher erfolglos die "Bauteilliste JLCB.ods" zum 1.4er Mainboard.
Hab sie versucht in KiCad zu exportieren, aber da kommt viel zu viel raus - und ohne LCSC.
Kannst du mir bitte sagen, wo ich die aktuelle Liste finde? Daanke und frohe Osterfeiertage/-ferien.
LG
Marcus

PS:
Ich weiß, es gibt einen neuen Controller und Treiber, aber die sind außerhalb meines Budgets ... und ich möchte es selbst bauen 😁