Ardumower/ardumower #143: extra mod: robomower + dog poop scooper

allanlaal, 2021-07-22:
what about adding a poop scooper to the ardumower for those of us, who own dogs? :)

something like this: https://www.youtube.com/watch?v=-40HjfJtOgo

## use case

I have a dog, and like most dogs its not trained to shit in a tiny square in the corner of my yard :open_mouth: 
whilst the lawmower will happily demolish the dried deposits, it still just throws it around that area - which will make some guests squemish

## solution

a custom Ardumower chassis with a poop scoop that when mowing:
 • will identify poop using a front camera, 
 • then will stop mowing
 • then turn around to face the crap pile with its scooper (which is a higher portion of the chassis than the mower blades):
[image]
 • then scoop the poop (no checks needed). I think that a robomowed lawn will not need a separate cutter like in the video - the torque of the motors with a sharpened scoop should be enough to grip and cut the grass picked up
 • it will go a deposit the poop on a predetermined area. maybe incrementing the location +10cm or similar to avoid running over the deposits :D 
 • then the Ardumower will resume mowing until it detects the next pile (or a previously half-scooped/failed pile)

allanlaal, 2021-07-22:
Q: "why not have 2 bots?
A: both would have to have an accurate coordinate system and a server to add jobs

flynneva, 2021-07-22:
maybe support a standard "attachment" connector? that way you could have multiple different attachments that the robot could hook up to?

im thinking like [tractor implements](https://www.google.com/search?q=tractor+implements&rlz=1C1GCEA_enUS892US892&source=lnms&tbm=isch&sa=X&ved=2ahUKEwjOzbelkPfxAhVQwZ4KHe0BBqMQ_AUoAnoECAEQBA&biw=2560&bih=1329&dpr=1)

flynneva, 2021-07-22:
also it could de-couple the development of the ardumower from the implements. so anyone could develop their own implement for it without adding additional burden to the devs of ardumower